srcfl/srcful-gateway #135: Test Helium BLE config

h0bb3, 2024-02-18:
Check if the nebra/rak/helium stuff can be configured using their ble apps.

(I had no luck so far)

h0bb3, 2024-02-18:
Using the app Helium Hotspot Utility you can connect but nothing seems to happen after that.

Other apps Helium Hotspot and Hotspotty Connect seems to require a Helium wallet

davmoz, 2024-04-07:
# Hotspot Utility

## Scenario:

1. Press BLE-button
2. Pair using Hotspot Utility
3. Wait a few minutes
4. Configure Wifi

## Outcome
Successful configuration of WiFi

[image]

### Normal operation & Button press logs
[logs-1.txt](https://github.com/srcfl/srcful-gateway/files/14897358/logs-1.txt)
[image]

### WiFi Configuration
[logs-2.txt](https://github.com/srcfl/srcful-gateway/files/14897370/logs-2.txt)
[image]
[image]

davmoz, 2024-04-07:
# Hotspot Utility

## Scenario:

1. Press BLE-button
2. Pair using Helium hotspot 
3. Configure WiFi

## Outcome
Unable to proceed due to firmware version

[image]

## Details
### Normal operation & Button press logs
```
2024-04-07T17:25:33+02:00  diagnostics  INFO:hw_diag.utilities.hardware:Retrieving list of BLE device(s)
2024-04-07T17:25:33+02:00  diagnostics  2024-04-07 15:25:33,437 - [INFO] - hw_diag.utilities.hardware - (hardware.py).get_ble_devices -- /opt/python-dependencies/lib/python3.11/site-packages/hw_diag/utilities/hardware.py:(106) - Retrieving list of BLE device(s)
2024-04-07T17:25:33+02:00  diagnostics  2024-04-07 15:25:33,449 - [INFO] - hw_diag.utilities.hardware - (hardware.py).get_ble_devices -- /opt/python-dependencies/lib/python3.11/site-packages/hw_diag/utilities/hardware.py:(129) - Found the following BLE Devices: [{'Address': 'E4:5F:01:37:0D:E1', 'Name': 'c04f2b3', 'Powered': '1', 'Discoverable': '0', 'Pairable': '0', 'Discovering': '0'}]
2024-04-07T17:25:33+02:00  diagnostics  INFO:hw_diag.utilities.hardware:Found the following BLE Devices: [{'Address': 'E4:5F:01:37:0D:E1', 'Name': 'c04f2b3', 'Powered': '1', 'Discoverable': '0', 'Pairable': '0', 'Discovering': '0'}]
2024-04-07T17:25:33+02:00  diagnostics  2024-04-07 15:25:33,449 - [INFO] - hw_diag.utilities.hardware - (hardware.py).get_lte_devices -- /opt/python-dependencies/lib/python3.11/site-packages/hw_diag/utilities/hardware.py:(175) - Retrieving list of LTE device(s)
2024-04-07T17:25:33+02:00  diagnostics  INFO:hw_diag.utilities.hardware:Retrieving list of LTE device(s)
2024-04-07T17:25:33+02:00  diagnostics  2024-04-07 15:25:33,457 - [INFO] - hw_diag.utilities.hardware - (hardware.py).get_lte_devices -- /opt/python-dependencies/lib/python3.11/site-packages/hw_diag/utilities/hardware.py:(202) - Found the following LTE Devices: []
2024-04-07T17:25:33+02:00  diagnostics  INFO:hw_diag.utilities.hardware:Found the following LTE Devices: []
...
2024-04-07T17:25:59+02:00  gateway-config  2024-04-07 15:25:59,601 - [DEBUG] - gatewayconfig.bluetooth.characteristics.firmware_revision_characteristic - (firmware_revision_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/firmware_revision_characteristic.py:(19) - Read Firmware
```
### WiFi Configuration
N/A

davmoz, 2024-04-07:
Another interesting thing is that name is not being fetched post BLE-pairing when using the Hotspot Utility app. 
```
2024-04-07T14:57:28+02:00  gateway-config  2024-04-07 12:57:28,828 - [ERROR] - gatewayconfig.bluetooth.characteristics.diagnostics_characteristic - (diagnostics_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/diagnostics_characteristic.py:(41) - Unexpected exception while trying to read diagnostics: org.freedesktop.DBus.Error.ServiceUnknown: The name com.helium.Miner was not provided by any .service files
``` 
Logs: [logs-3.txt](https://github.com/srcfl/srcful-gateway/files/14897807/logs-3.txt)

davmoz, 2024-04-07:
Also tested RAK's own [app](https://apps.apple.com/us/app/mntd-helium-hotspot-manager/id1595569593) for onboarding and setup of MNTD. and RAK Hotspots for the Helium Network, which doesn't seem to play well with Nebras. 

[image]
[image]

Logs during this period: 
```
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,424 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(27) - Read WiFi Services
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,656 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(36) - Inspecting SSID iotlab
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,658 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(36) - Inspecting SSID iotlab2
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,657 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(39) - iotlab is a valid ssid
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,657 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(43) - iotlab ssid is unknown
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,658 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(39) - iotlab2 is a valid ssid
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,658 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(43) - iotlab2 ssid is unknown
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,658 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(36) - Inspecting SSID iotlab2_5G
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,658 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(39) - iotlab2_5G is a valid ssid
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,658 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(43) - iotlab2_5G ssid is unknown
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,658 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(36) - Inspecting SSID Orion_5G
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,659 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(39) - Orion_5G is a valid ssid
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,659 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(43) - Orion_5G ssid is unknown
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,659 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(36) - Inspecting SSID TN_24GHz_F55D95
2024-04-07T17:54:59+02:00  gateway-config  2024-04-07 15:54:59,659 - [DEBUG] - gatewayconfig.bluetooth.characteristics.wifi_services_characteristic - (wifi_services_characteristic.py).ReadValue -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/bluetooth/characteristics/wifi_services_characteristic.py:(39) - TN_24GHz_F55D95 is a valid ssid
[... cut ...] /venv/lib/python3.9/site-packages/gatewayconfig/processors/bluetooth_advertisement_processor.py:(43) - Stopping Bluetooth advertisement
2024-04-07T17:56:26+02:00  gateway-config  GATT advertisement UNregistered
2024-04-07T17:56:29+02:00  gateway-config  2024-04-07 15:56:29,632 - [DEBUG] - gatewayconfig.processors.diagnostics_processor - (diagnostics_processor.py).run -- /opt/venv/lib/python3.9/site-packages/gatewayconfig/processors/diagnostics_processor.py:(55) - Running DiagnosticsProcessor
```

h0bb3, 2024-04-25:
Basically BLE for the nebra stuff does not seem to work with our current dual mining firmware.

closed 2024-04-25 by h0bb3