srcfl/srcful-gateway #143: Test RPi Zero

h0bb3, 2024-03-11:
Are there any potential issues we need to test for the RPi Zero?

The button code for Ble maybe.

Would it be a good idea to incorporate a button at this point?

h0bb3, 2024-03-11:
Yes to button

I guess we need a special fleet for this hw.

davmoz, 2024-03-11:
> Yes to button
> 
> I guess we need a special fleet for this hw.

Agree, and we already have a fleet and one device up and running. A second one joining this evening. Will take a look into how we can incorporate a button.

h0bb3, 2024-03-11:
There is code for the button already in the ble component, you just supply a pin via a command line argument (in practice an env variable) Check the rak compose file for an example.

Maybe we need to rethink this and make it more intelligent in some way to avoid having fleets with/without buttons... i.e. the original with/without button code. I.e. it could possibly be some setting that you can do via the configurator instead.

Maybe this could be something to think about in a ble behavior setting.

I.e. advertise always, advertise on button, advertise on power cycle. Maybe also set the advertise time...

davmoz, 2024-03-28:
Seems to be working, but lets not consider this now

closed 2024-03-28 by davmoz