srcfl/srcful-gateway #148: Variable harvest registers

h0bb3, 2024-03-21:
It would be more optimal to just read everything once per harvest-transport and then just a minimal set of registers. This would send less data over the network and also increase performance of the inverter connection.

This should not be hard to do i the Harvester - but the inverter configurations needs to have the minimal and large set of registers.

The Inverter interface could be changed to full_harvest and mini_harvest

davmoz, 2024-03-28:
Proposal: 

1. Full read - All registers
2. Minimal read - essential registers (could be minimal read or full read depending on performance, and is the data sent to backend) 
3. Filter - Necessary registers

h0bb3, 2024-04-25:
I think this is done Why no assignee and no status set?! @davmoz

closed 2024-04-25 by h0bb3