srcfl/srcful-gateway #149: reading inverter registers performance on Huawei

h0bb3, 2024-03-22:
Reading the registers take too long on e.g. Huawei this blocks the task queue and makes everything unresponsive.

The assumption has always been to not block the message queue and use threading if there is a risk of long delays.

Harvest now must be threaded or registers removed from Huawei.

```
[Logs]    [2024-03-22T20:09:12.096Z] [web] DEBUG:server.tasks.harvest:Harvest took 13856 ms
[Logs]    [2024-03-22T20:09:44.490Z] [web] DEBUG:server.tasks.harvest:Harvest took 17125 ms
[Logs]    [2024-03-22T20:10:16.370Z] [web] DEBUG:server.tasks.harvest:Harvest took 14754 ms
[Logs]    [2024-03-22T20:10:47.493Z] [web] DEBUG:server.tasks.harvest:Harvest took 15051 ms
[Logs]    [2024-03-22T20:11:16.799Z] [web] DEBUG:server.tasks.harvest:Harvest took 14256 ms
[Logs]    [2024-03-22T20:11:46.509Z] [web] DEBUG:server.tasks.harvest:Harvest took 15454 ms
[Logs]    [2024-03-22T20:12:17.184Z] [web] DEBUG:server.tasks.harvest:Harvest took 15221 ms
[Logs]    [2024-03-22T20:12:47.252Z] [web] DEBUG:server.tasks.harvest:Harvest took 14847 ms
[Logs]    [2024-03-22T20:13:19.754Z] [web] DEBUG:server.tasks.harvest:Harvest took 17654 ms
[Logs]    [2024-03-22T20:13:52.233Z] [web] DEBUG:server.tasks.harvest:Harvest took 14826 ms
```

This needs to be patched asap!

h0bb3, 2024-03-22:
Another idea is to thread the queue itself... then we do not risk problems like this and it will likely be simpler code in the tasks... but this will take some time to implement - so patch first :)

davmoz, 2024-03-23:
Improved in 028cda6cdb1d01dbfab06d8b271d19e3d5196d4f

closed 2024-03-23 by davmoz