srcfl/srcful-gateway #152: Unexpected uptime

davmoz, 2024-03-25:
After being switched off for two days, a device, upon being powered back on and connected to a new WiFi access point, displayed an unexpected uptime of more than 57 hours.

h0bb3, 2024-03-27:
Very strange... the uptime is handled by simply setting the start time to when the blackboard is initialized.

in the blackboard constructor: `self._start_time = self.time_ms()`

at the then in the api: {"msek": `data.bb.time_ms() - data.bb.start_time}`

```
def time_ms(self):
        return time.time_ns() // 1_000_000
```

so possible causes could be - large jump in time.time_ns() caused by some new time being set due to network config, in some cases time can be updated via network syncronization I guess?

somewere in the code _start_time gets manipulated (I doubt this)

h0bb3, 2024-03-31:
moved to  `time.monotonic_ns()` that is not affected by external sources for time adjustment.

h0bb3, 2024-04-05:
monotonic time is now only used to get the elapsed time. Also added test cases for the time methods in the blackboard.

closed 2024-04-05 by h0bb3