srcfl/srcful-gateway #174: Feature request. MODBUS Proxy

frahlg, 2024-05-10:
I would like to add some complexity to the GW, I hope this could be added as it's own service and hopefully not interact with too much of the other complexity. It's configured via a .yaml file so instead we need to push settings to the proxy instead of the GW. and then the modbus proxy interacts directly.

OR an alternative could be to build this into our FW directly instead of having as a separate service.

Why this is needed?

- In examples where customers have other services running, such as 1,5 Hearbeat, Checkwatt or maybe their own Home Assistant service - we need to point these to the proxy and all can work in tandem.
- From the security perspective we can also say that we have a built in FW/proxy for the modbus.
- Added bonus, which might be the most important for future development: we will also be able to scrape all existing communication that happens from Heartbeat, Checkwatt etc.

I have been running this months and it seems to be very stable. Example below.

$ modbus-proxy -c ./modbus-config.yml

https://github.com/tiagocoutinho/modbus-proxy

h0bb3, 2024-05-10:
Connected to #45

h0bb3, 2024-05-11:
duplicate of #161 (closing that one as this has some more info)

h0bb3, 2024-05-13:
Also note licence information for used lib/code/software and how that would affect the eGW

It seems a modbus proxy is basically 400 lines of code... is it really worth launching this as a service of it's own?

https://github.com/tiagocoutinho/modbus-proxy/blob/master/src/modbus_proxy.py

erikarenhill, 2024-05-23:
It would be nice if the srcful proxy would be able to cache the last response available for a certain register, for example lets say an inverter can only respond every 5th second, if some other service like a Home Assistant addon would request data every second from the srcful proxy it would be better to just respond with the last known value instead of pushing the inverter harder? Just some thoughts

h0bb3, 2024-07-08:
if we want to support multiple inverters this also needs to be sorted.