srcfl/srcful-gateway #175: Monitor cryptochip revive

davmoz, 2024-05-20:
We need to understand if revive works and how many time it takes to actually fix the zeros pubkey issue. 

**End goal:** To proactivley run revive to avoid problems.

davmoz, 2024-05-20:
Scenario:
* {"exception": "Failed to get public key cryptauthlib error code: 250", "endpoint": "/api/crypto"}
* configurator kraschar pga pubkey parsing (undefined, så crypto döden) 
* Revive 2 gånger - http://192.168.50.246/api/crypto/revive 
* http://192.168.50.246/api/crypto/revive
* Working

h0bb3, 2024-05-21:
Several eGWs went offline this night. All are revved:
9694f57: x2
64982ae: x1
5e43744: x1

h0bb3, 2024-05-23:
Offline due to crypto:
c3712ed: x1

h0bb3, 2024-05-31:
interesting case that took two attempts. The first attempt returned:

64982ae

```
"status": "atcab_get_pubkey: 250 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000\nfinal return value (0 == success): 250\n"}
```

so actually another error code than the usual

h0bb3, 2024-06-04:
64982ae

revived again on first attempt

h0bb3, 2024-06-07:
c3712ed

revived on first attempt

h0bb3, 2024-06-10:
64982ae

revived on first attempt

h0bb3, 2024-06-10:
9694f57

revived on second attempt

h0bb3, 2024-06-17:
Manual revive works so I think we close this one.

we had 10 revives done during these 4 weeks. Now we need to evaluate the automatic revive.

closed 2024-06-17 by h0bb3