srcfl/srcful-gateway #176: Test nmap

davmoz, 2024-05-20:
test the implemented nmap endpoint. 

End goal: implement this in the app.

davmoz, 2024-05-21:
seems to be working

h0bb3, 2024-05-23:
Tested on a site that had lost connection to inverter. After scanning the new ip was shown and I could connect to the inverter using this ip address instead.

h0bb3, 2024-05-24:
Another site that had lost connection to inverter due to change ip. After scanning the new ip was shown and I could connect to the inverter using this ip address instead.

davmoz, 2024-06-01:
This is now available from: 

Gateway: 7439f32bd968d5d100e1e35febe989150aeee9a4 
App: [f00cc588fb2a598441c898ca7fb6ea5813c263c0](https://github.com/srcfl/srcful-configurator/commit/f00cc588fb2a598441c898ca7fb6ea5813c263c0)

closed 2024-06-01 by davmoz