srcfl/srcful-gateway #188: Helium Wallet Compatibility Prototype

h0bb3, 2024-05-28:
The helium wallet app seems to simply use a number of characteristics for connecting to the hotspot. To be compatible we can just fulfill these

https://docs.helium.com/hotspot-makers/become-a-maker/hotspot-ble-services/

I'm not sure what everything is here and possibly the wallet connection is not there? Maybe that is handled in som other way in Helium?

There seems to be an erlang implementation here: https://github.com/helium/gateway-config

Lets make a prototype to test this.

h0bb3, 2024-05-28:
If we become compatible here we would essentially not need our own app if we can handle the custom configuration stuff via web app.

h0bb3, 2024-05-28:
Good progress. I can now communicate with the helium app with my little hard coded prototype.
There is a high probability that this actually will work with bless also. This would be good as we already have a container up and running for this.

There are some parts where we would need some work e.g information for some endpoints and also looking for what actually happens in the different steps when writing data.

The app uses google protocol buffers so this lib needs to be installed. I will try with bless again tomorrow also.

h0bb3, 2024-05-28:
At this point I don't see it as impossible to integrate the Helium wallet to the next release. This would then likely work for setting wifi, position, and wallet initialization.

h0bb3, 2024-05-29:
An add gateway seems to be sent immediately when the gateway is connected in the app. The reply should conform to:

https://github.com/NebraLtd/hm-pyhelper/issues/91

h0bb3, 2024-06-01:
To a large degree working with hard coded endpoints. The main issue now is that the Device Information Service is published twice (once by OS?), and then seems to not be picked up by the app.

I have tried libraries bless, and raw dbus stuff (linux only) as well as nebras firmware. Always the same result. I will try running this on an actual rpi - but I cannot relly see how this would change the situation.

I have also tried to list curent services but this always returns nothing so it does not seem to be there from the start but rather that it is added by default.

h0bb3, 2024-06-01:
This issue seems to have been lifted in bluez and also referenced by the helium app. Will investigate further.

https://github.com/bluez/bluez/issues/101

h0bb3, 2024-06-01:
Setting `DeviceID = false` seems to fix this issue. This means we need to start the ble daemon with a cusom file. WIll make issue for this.

h0bb3, 2024-06-01:
Ok some good progress. We are now stuck at 'no manufacturer for this hotspot' I suspect that there is some check that the hotspot is actually valid. I'm guessing the hotspot serial or pubkey is used to check that it is an acutal valid hotspot. The suggested values from the docs does not seem to worl.

But the Hotspot Util App finally started showing some info. Next step would be to get some actual information and action (i.e. wifi settings to work)

davmoz, 2024-06-01:
> Ok some good progress. We are now stuck at 'no manufacturer for this hotspot' I suspect that there is some check that the hotspot is actually valid. I'm guessing the hotspot serial or pubkey is used [...]

This might also be displayed due to insufficient funds (SOL).

h0bb3, 2024-06-02:
Ah ok - yes I have zero... Maybe I will switch the wallet to testnet and get some funds into it.

h0bb3, 2024-06-02:
There is a large chance that we have come as far as we can without having a maker account.

https://docs.helium.com/hotspot-makers/become-a-maker/hotspot-integration-testing/#adding-hotspots-to-the-onboarding-server

frahlg, 2024-06-02:
We will bring this up with Joey today. /Fredrik Ahlgren

https://docs.helium.com/hotspot-makers/become-a-maker/hotspot-integration-testing/#adding-hotspots-to-the-onboarding-server

h0bb3, 2024-06-02:
Some info from Joey: https://github.com/helium/gateway-config

h0bb3, 2024-06-03:
some more work done. It seems both our original service and the helium service need to be the single service on the device (I find this a bit strange - but if both are primary services the either out app or the helium wallet app do not work)

We may need to merge our chars into the helium service and this would be a major breaking change in the app (we can manage this though).

We also may need to override the bless lib so that it can create multiple primary services (to manage the DIS service)...

davmoz, 2024-10-10:
Most of this is implemented in the [dual-mining](https://github.com/srcfl/srcful-gateway/tree/dual-mining) branch. Planned to be rolled out Nov 6th.

davmoz, 2024-10-31:
this is done and working in main, closing.

closed 2024-10-31 by davmoz