srcfl/srcful-gateway #190: feature request: add api endpoint to get the serial of the gateway

erikarenhill, 2024-05-30:
as someone might need the serial of the gateway as input to some energy-api / graphql requests, it would be nice if there would be a json endpoint for fetching this. From what I can tell it's only presented in the root page through the html right now?

I propose a common endpoint for gateway data, such as making a single call to fetch multiple stuff that doesn't change very often
for example

GET /api/gateway

returns
{
   name,
   version,
   serial
}

h0bb3, 2024-05-30:
/api/crypto is your friend

check

/doc for full api docs 😁

erikarenhill, 2024-05-30:
> /api/crypto is your friend
> 
> check
> 
> /doc for full api docs 😁

oh you mean the one with documented null response, already checked /doc but didnt try this one out :) 
```
   "GET": {
      "crypto": null,
```

I suppose it's updated in next firmware, didn't check the unreleased code :)

Thanks for pointing me in the right direction!

closed 2024-05-30 by erikarenhill

h0bb3, 2024-05-31:
Yes, there was a bug in the docs for that one but it is actually there :) (and this will be fixed in the next release)