srcfl/srcful-gateway #193: read ac power in addition to dc power

davmoz, 2024-06-04:
A deviation exists between current power readings from our energy API and local Modbus TCP readings. Address this and consider changing to AC power (or add it) as it is more interesting from the grid point of view.

h0bb3, 2024-06-04:
I think both makes sense to collect as this information can be interesting for e.g. customers.

h0bb3, 2024-06-24:
The main thing why we did not use AC was that the power was connected to (old) tokenomics and different inverters reported this using different models derrived from DC,  actual measurements,  or in some cases not at all (?)

davmoz, 2024-10-10:
SunSpec readings now include AC and DC power. We create separate issues for each inverter model when using Modbus-only readings.

closed 2024-10-10 by davmoz