srcfl/srcful-gateway #210: Inverter Identity

h0bb3, 2024-07-18:
Get the serial or some other identity for the inverter when connected. Save this as part of the bootstrap/settings. This will allow the connection/reconnection/scan logic to know what exact inverter to connect to and to avoid connecting to the wrong inverter if there are several inverters on the network.

I.e. it will be part of the reconnection process to check that it is actually the correct inverter.

h0bb3, 2024-07-22:
if we combine e.g. inverter serial and mac address we should be gtg.

This would in theory eliminate the tricky situation of having a proxy and an inverter on the same net (currently the inverter may show up first when the connection should be to the proxy)

h0bb3, 2024-08-12:
It is likely a bit tricky to get the mac adresse of a remote host from within a docker container. Esp as we are not running in host mode.

I would think https://pypi.org/project/scapy/ is the most promising approach if not simply elevating the container network mode.

h0bb3, 2024-09-06:
I think we need this one asap esp if we are soon supporting multiple inverters, e.g. we need to add the id to the jwt header so we can know what device was harvested.

davmoz, 2024-10-14:
Conditional reconnect post-scan was implemented in 6973efc55321114d683ab9a35bc0ba6299d033a9, see https://github.com/srcfl/srcful-gateway/issues/214#issuecomment-2411475358. 

This is partially implemented in b91a161a3eec9eeb296f594e4d8dcab3a00b31f1. Now there is a new field in the header, `sn`. **Solarman** and **SunSpec** readings will have a valid `sn`, while **Modbus TCP** will have the `MAC address` of the device and **Modbus RTU** `N/A`. 

I suggest we create a new issue for each model, and close this one for now.

closed 2024-10-14 by davmoz