srcfl/srcful-gateway #217: Configurable backend URL

kneeclass, 2024-08-16:
We need to implement a mechanism that allows the backend URL to be changeable for each device through a configuration process. This will enable easier testing, development, and deployment across different environments and configurations.

h0bb3, 2024-08-18:
I think this should be part of the settings for the gateway. I would guess that there could be a need to change this based on service provider also (if there are multiple SPs).

We can of course make a test asap on a separate fleet.

h0bb3, 2024-08-19:
Add support for sending to multiple endpoints. This can be good for debugging testing, without offlining the gateway.

h0bb3, 2024-08-21:
This seems to be up and running now. Could change to mainnet for https://explorer.srcful.io/egw/0123f863dd5d878dee

There are GET and POST endpoints for the settings. Currently no checks of validity etc. so it is rather raw and crude.

Settings are not persistent atm, as this will be a backend thing.

All of this wil be in other issues though so this one is closed.

closed 2024-08-21 by h0bb3