srcfl/srcful-gateway #235: JSON Serialization Fails Due to Unexpected Socket Object in Configuration Data

davmoz, 2024-09-11:
Running latest main https://github.com/srcfl/srcful-gateway/commit/8abce9056cc51c4c6891c9ba23c4153c66285ced 
```
[Logs]    [2024-09-11T06:48:11.882Z] [web] ERROR:server.tasks.srcfulAPICallTask:Error in SrcfulAPICallTask Object of type socket is not JSON serializable'Traceback (most recent call last):\n  File "/server/tasks/srcfulAPICallTask.py", line 64, in execute\n    response = post()\n               ^^^^^^\n  File "/server/tasks/srcfulAPICallTask.py", line 55, in post\n    json = self._json()\n           ^^^^^^^^^^^^\n  File "/server/tasks/configurationMutationTask.py", line 37, in _json\n    jwt = self._build_jwt()\n          ^^^^^^^^^^^^^^^^^\n  File "/server/tasks/configurationMutationTask.py", line 20, in _build_jwt\n    message = crypto.jwtlify(self.data)\n              ^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/server/crypto/crypto.py", line 92, in jwtlify\n    return base64_url_encode(json.dumps(data).encode("utf-8")).decode("utf-8")\n                             ^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/__init__.py", line 231, in dumps\n    return _default_encoder.encode(obj)\n           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 200, in encode\n    chunks = self.iterencode(o, _one_shot=True)\n             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 258, in iterencode\n    return _iterencode(o, 0)\n           ^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 180, in default\n    raise TypeError(f\'Object of type {o.__class__.__name__} \'\nTypeError: Object of type socket is not JSON serializable'
[Logs]    [2024-09-11T06:48:11.884Z] [web] WARNING:server.tasks.configurationMutationTask:Failed to mutate configuration state
[Logs]    [2024-09-11T06:48:12.132Z] [web] ERROR:server.tasks.srcfulAPICallTask:Error in SrcfulAPICallTask Object of type socket is not JSON serializable'Traceback (most recent call last):\n  File "/server/tasks/srcfulAPICallTask.py", line 64, in execute\n    response = post()\n               ^^^^^^\n  File "/server/tasks/srcfulAPICallTask.py", line 55, in post\n    json = self._json()\n           ^^^^^^^^^^^^\n  File "/server/tasks/configurationMutationTask.py", line 37, in _json\n    jwt = self._build_jwt()\n          ^^^^^^^^^^^^^^^^^\n  File "/server/tasks/configurationMutationTask.py", line 20, in _build_jwt\n    message = crypto.jwtlify(self.data)\n              ^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/server/crypto/crypto.py", line 92, in jwtlify\n    return base64_url_encode(json.dumps(data).encode("utf-8")).decode("utf-8")\n                             ^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/__init__.py", line 231, in dumps\n    return _default_encoder.encode(obj)\n           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 200, in encode\n    chunks = self.iterencode(o, _one_shot=True)\n             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 258, in iterencode\n    return _iterencode(o, 0)\n           ^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 180, in default\n    raise TypeError(f\'Object of type {o.__class__.__name__} \'\nTypeError: Object of type socket is not JSON serializable'
[Logs]    [2024-09-11T06:48:12.134Z] [web] WARNING:server.tasks.configurationMutationTask:Failed to mutate configuration state
```

h0bb3, 2024-09-11:
gw: 4690e4b

h0bb3, 2024-09-11:
No idea where this comes from but have added better logging in the SaveStateTask now at least...

davmoz, 2024-09-11:
```
[Logs]    [2024-09-11T14:17:43.676Z] [web] DEBUG:server.web.socket.settings_subscription:Received pong: b''
[Logs]    [2024-09-11T14:17:43.998Z] [web] DEBUG:server.tasks.harvest:Harvest took 7 ms
[Logs]    [2024-09-11T14:17:44.999Z] [web] DEBUG:server.tasks.harvest:Harvest took 7 ms
[Logs]    [2024-09-11T14:17:46.001Z] [web] DEBUG:server.tasks.harvest:Harvest took 8 ms
[Logs]    [2024-09-11T14:17:46.003Z] [web] INFO:server.tasks.harvest:Creating transport for https://mainnet.srcful.dev/gw/data/
[Logs]    [2024-09-11T14:17:46.296Z] [web] ERROR:server.tasks.srcfulAPICallTask:Error in SrcfulAPICallTask Object of type socket is not JSON serializable'Traceback (most recent call last):\n  File "/server/tasks/srcfulAPICallTask.py", line 64, in execute\n    response = post()\n               ^^^^^^\n  File "/server/tasks/srcfulAPICallTask.py", line 55, in post\n    json = self._json()\n           ^^^^^^^^^^^^\n  File "/server/tasks/configurationMutationTask.py", line 37, in _json\n    jwt = self._build_jwt()\n          ^^^^^^^^^^^^^^^^^\n  File "/server/tasks/configurationMutationTask.py", line 20, in _build_jwt\n    message = crypto.jwtlify(self.data)\n              ^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/server/crypto/crypto.py", line 92, in jwtlify\n    return base64_url_encode(json.dumps(data).encode("utf-8")).decode("utf-8")\n                             ^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/__init__.py", line 231, in dumps\n    return _default_encoder.encode(obj)\n           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 200, in encode\n    chunks = self.iterencode(o, _one_shot=True)\n             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 258, in iterencode\n    return _iterencode(o, 0)\n           ^^^^^^^^^^^^^^^^^\n  File "/usr/local/lib/python3.11/json/encoder.py", line 180, in default\n    raise TypeError(f\'Object of type {o.__class__.__name__} \'\nTypeError: Object of type socket is not JSON serializable'
[Logs]    [2024-09-11T14:17:46.298Z] [web] ERROR:server.tasks.saveStateTask:Failed to save state <Response [None]>
[Logs]    [2024-09-11T14:17:46.299Z] [web] ERROR:server.tasks.saveStateTask:State: {'status': {'version': '0.14.1', 'uptime': 422962, 'messages': [{'message': "Device opened: {'connection': 'SUNSPEC', 'host': '192.168.1.10', 'port': 502, 'address': 1}", 'type': 'info', 'timestamp': 1726063785, 'id': 68}]}, 'crypto': {'deviceName': 'ATECC608', 'serialNumber': '0123f863dd5d878dee', 'publicKey': 'ab7eb60036ffd043641d4639822fcfeab2a9952d6fe41158347a823681fce90d352149e1f520c4f7e94e1802646e605781901c1a792da38807a9bf4be7e2859a', 'compactKey': '112JXcJfm53q4CgEnNvuizAjkDKmPjKvzgUodq5WzbY65Ghfr3JE', 'chipDeathCount': 0}, 'network': {'wifi': {'ssids': ['fredrik.madeleine']}, 'address': {'ip': dbus.String('192.168.1.212', variant_level=1), 'port': 80, 'eth0_mac': dbus.String('DC:A6:32:F8:38:F7', variant_level=1), 'wlan0_mac': dbus.String('DC:A6:32:F8:38:FA', variant_level=1), 'interfaces': {dbus.String('wlan0', variant_level=1): dbus.String('192.168.1.212', variant_level=1), dbus.String('supervisor0', variant_level=1): dbus.String('10.114.104.1', variant_level=1)}}}, 'devices': {'configured': [{'connection': {'connection': 'SUNSPEC', 'host': '192.168.1.10', 'port': 502, 'address': 1}, 'is_open': <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.18.0.2', 53526), raddr=('192.168.1.10', 502)>}], 'supported': {'inverters': [{'name': 'huawei', 'dname': 'Huawei', 'proto': 'mb'}, {'name': 'solaredge', 'dname': 'SolarEdge', 'proto': 'mb'}, {'name': 'sungrow', 'dname': 'Sungrow', 'proto': 'mb'}, {'name': 'sungrow_hybrid', 'dname': 'Sungrow Hybrid', 'proto': 'mb'}, {'name': 'sma', 'dname': 'SMA', 'proto': 'mb'}, {'name': 'fronius', 'dname': 'Fronius', 'proto': 'mb'}, {'name': 'fronius_sunspec', 'dname': 'Fronius (SunSpec)', 'proto': 'mb'}, {'name': 'deye', 'dname': 'Deye', 'proto': 'sol'}, {'name': 'deye_hybrid', 'dname': 'Deye Hybrid', 'proto': 'sol'}, {'name': 'growatt', 'dname': 'Growatt', 'proto': 'mb'}, {'name': 'goodwe', 'dname': 'GoodWe', 'proto': 'mb'}, {'name': 'lqt40s', 'dname': 'LQT40S', 'proto': 'mb'}, {'name': 'ferroamp', 'dname': 'ferroamp', 'proto': 'mb'}]}}}
[Logs]    [2024-09-11T14:17:46.299Z] [web] WARNING:server.tasks.configurationMutationTask:Failed to mutate configuration state
[Logs]    [2024-09-11T14:17:46.312Z] [web] DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): mainnet.srcful.dev:443
[Logs]    [2024-09-11T14:17:46.378Z] [web] DEBUG:urllib3.connectionpool:https://mainnet.srcful.dev:443 "POST /gw/data/ HTTP/11" 200 0
[Logs]    [2024-09-11T14:17:46.380Z] [web] INFO:server.tasks.harvestTransport:Response: <Response [200]>
[Logs]    [2024-09-11T14:17:46.441Z] [web] DEBUG:server.network.wifi:network device path: /org/freedesktop/NetworkManager/Devices/1
[Logs]    [2024-09-11T14:17:46.448Z] [web] DEBUG:server.network.wifi:device state 100
[Logs]    [2024-09-11T14:17:46.458Z] [web] DEBUG:server.network.wifi:device ip address 127.0.0.1
[Logs]    [2024-09-11T14:17:46.459Z] [web] DEBUG:server.network.wifi:network device path: /org/freedesktop/NetworkManager/Devices/2
[Logs]    [2024-09-11T14:17:46.468Z] [web] DEBUG:server.network.wifi:device state 20
[Logs]    [2024-09-11T14:17:46.469Z] [web] DEBUG:server.network.wifi:network device path: /org/freedesktop/NetworkManager/Devices/3
[... cut ...] to': 'sol'}, {'name': 'growatt', 'dname': 'Growatt', 'proto': 'mb'}, {'name': 'goodwe', 'dname': 'GoodWe', 'proto': 'mb'}, {'name': 'lqt40s', 'dname': 'LQT40S', 'proto': 'mb'}, {'name': 'ferroamp', 'dname': 'ferroamp', 'proto': 'mb'}]}}}
[Logs]    [2024-09-11T14:17:59.085Z] [web] WARNING:server.tasks.configurationMutationTask:Failed to mutate configuration state
[Logs]    [2024-09-11T14:17:59.222Z] [web] DEBUG:server.network.wifi:network device path: /org/freedesktop/NetworkManager/Devices/1
```