srcfl/srcful-gateway #278: Configuration Migration Strategy for Device Settings

davmoz, 2024-11-05:
When the gateway starts up and synchronizes device settings from the cloud, it causes an issue with legacy configurations. 

These older settings may lack required fields that were introduced in newer firmware versions. Currently, these incomplete configurations aren't being removed or updated, which results in the gateway creating new OpenPerpetualDeviceTasks for potentially invalid device configurations.

Configuration formats will likely continue to evolve with future firmware updates. We need a systematic approach to handle version differences between stored configurations and current firmware requirements.

h0bb3, 2024-11-20:
Json is flexible as you can always add information - and then deprecate the old field.

I suggest this is handled as close as possible to the actual definition of the setting in question.

h0bb3, 2024-11-27:
This should basically be handled in the `SettingsDeviceListener` at a high level. Then in the respective device class at a low level.

davmoz, 2025-01-09:
current format seems to be robust enough to handle such changes in the future. closing

closed 2025-01-09 by davmoz