srcfl/srcful-gateway #299: Refactor queue in app.py

davmoz, 2024-11-11:
Partially related to https://github.com/srcfl/srcful-gateway/issues/237, but specifically refactor the queue in app.py. 

Would be cool to have also have an endpoint that returns all the tasks in the queue.

h0bb3, 2024-11-15:
Not really sure what the goal of the refactoring is here. But I will try to put the task queue in the blackboard, then added tasks to the blackboard would be directly added to the queue.

There could however be threading trolls here...

h0bb3, 2024-11-15:
This is done. Getting the tasks form the queue is impossible without destroying the queue or implementing a cached list of tasks. One alternative could be to implement a task observer thing that can be used - but imho this is overkill atm.

Decupled the blackboard from the TaskScheduler by introducing two interfaces SystemTime and TaskSource. Blackboard implements these two interfaces. Could be interesting to look into refactoring parts that do not need to depend on the entire blackboard.