srcfl/srcful-gateway #305: Improve WiFi-scanning...

davmoz, 2024-11-15:
Implement exponential backoff for the scanning frequency. Initially scan rapidly, like once every 5 seconds, then gradually reduce frequency until reaching a longer interval (like 2 minutes or something). Allow users to manually trigger a scan which resets back to rapid scanning.

h0bb3, 2024-11-15:
start the scan immediately in the app not after 45 seconds. Run with exponential backoff until you get a result.

h0bb3, 2024-11-26:
we now start scanning after 10 seconds and if we find a result then we send the state every 10 seconds for 30 seconds.

if the scan returns no result after 10 seconds we scan again (this is done max 3 times)

h0bb3, 2024-11-26:
This is done but needs testing on real gateway.

h0bb3, 2025-01-01:
Scanning still seems dodgy and does not set the state properly? at least not with the GET endpoint #341 but closing this one.

closed 2025-01-01 by h0bb3