srcfl/srcful-gateway #368: SN - Move from MAC to actual device serial numbers.

davmoz, 2025-01-31:
While SolarmanV5 sticks and SunSpec compatible devices have an easy to fetch SN, it needs to be assembled from multiple registers when reading using regular modbus tcp. 

Now that we have definitions for ~12 makes, we should consider moving to this and using MAC as a temporarily fallback SN until the definitions for the new make are in place.  

Perhaps we should keep the MAC in the header, with a new key.

h0bb3, 2025-02-05:
Do this for one model at a time, maybe as sub issues.

davmoz, 2025-02-12:
Seems like some inverters do not specify SN registers in their documentation, so these will be using their MAC Address as SN. 

Completed and available in 4d3fe598eda9cb19c2100d22d2c3d05851a551fd

closed 2025-02-12 by davmoz